qiskit-community/qiskit-metal #1003: LOM - Capacitive matrix not right size

FactOxx, 2024-11-28:
### Information

- **Qiskit Metal version**:  0.1.5
- **Python version**: 3.10
- **Operating system**: Windows 11

### What is the current behavior?
while running the simulation on ansys, it gives the error of "{
	"name": "ValueError",
	"message": "Capacitance matrix is not the right size",
	"stack": "---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[33], line 4
      1 c1.setup.junctions=Dict(Lj=10, Cj=0)
----> 4 c1.run_lom()
      5 c1.lumped_oscillator_all

File ~\\AppData\\Local\\Packages\\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\\LocalCache\\local-packages\\Python310\\site-packages\\qiskit_metal\\analyses\\quantization\\lumped_oscillator_model.py:171, in LOManalysis.run_lom(self)
    169 all_res = {}
    170 for idx_cmat, df_cmat in self.sim.capacitance_all_passes.items():
--> 171     res = extract_transmon_coupled_Noscillator(
    172         df_cmat,
    173         ic_amps,
    174         cj,
    175         num_cpads,
    176         fbus,
    177         fread,
    178         g_scale=1,
    179         print_info=bool(
    180             idx_cmat == len(self.sim.capacitance_all_passes)))
    181     all_res[idx_cmat] = res
    182 self.lumped_oscillator = all_res[len(self.sim.capacitance_all_passes)]

File ~\\AppData\\Local\\Packages\\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\\LocalCache\\local-packages\\Python310\\site-packages\\qiskit_metal\\analyses\\quantization\\lumped_capacitive.py:198, in extract_transmon_coupled_Noscillator(capMatrix, Ic, CJ, N, fb, fr, res_L4_corr, g_scale, print_info)
    196     raise ValueError('N must positive')
    197 if len(capMatrix) != (N + 3):
--> 198     raise ValueError('Capacitance matrix is not the right size')
    200 # make list of angular frequencies of resonators
    201 wr = np.zeros(N)  # angular freq of resonators (GHz-rad)

ValueError: Capacitance matrix is not the right size"
}"

### Steps to reproduce the problem

Run this simple code for the LOM analysis import qiskit_metal as metal
from qiskit_metal import designs, draw
from qiskit_metal import MetalGUI, Dict, Headings
import pyEPR as epr

design = designs.DesignPlanar({}, True)
design.chips.main.size['size_x'] = '2mm'
design.chips.main.size['size_y'] = '2mm'

gui = MetalGUI(design)

from qiskit_metal.qlibrary.tlines.meandered import RouteMeander
from qiskit_metal.qlibrary.terminations.open_to_ground import OpenToGround
from qiskit_metal.qlibrary.terminations.short_to_ground import ShortToGround

design.delete_all_components()

otg = OpenToGround(design, 'open_to_ground', options=dict(pos_x='1.25mm',  pos_y='0um', orientation='0'))
stg = ShortToGround(design, 'short_to_ground', options=dict(pos_x='-1.25mm',  pos_y='0um', orientation='180'))
rt_meander = RouteMeander(design, 'readout',  Dict(
        total_length='6 mm',
        hfss_wire_bonds = True,
        fillet='90 um',
        lead = dict(start_straight='100um'),
        pin_inputs=Dict(
            start_pin=Dict(component='short_to_ground', pin='short'),
            end_pin=Dict(component='open_to_ground', pin='open')), ))

gui.rebuild()
gui.autoscale()

gui.screenshot()

from qiskit_metal.analyses.quantization import LOManalysis
c1 = LOManalysis(design, "q3d")

q3d = c1.sim.renderer
q3d.start()

q3d.activate_ansys_design("TransmonQubit_q3d", 'capacitive')

q3d.render_design(['readout'], [])
q3d.analyze_setup("Setup")

c1.sim.capacitance_matrix, c1.sim.units = q3d.get_capacitance_matrix()
c1.sim.capacitance_all_passes, _ = q3d.get_capacitance_all_passes()
print(c1.sim.capacitance_matrix)

c1.run_lom()
print(c1.lumped_oscillator_all)

c1.plot_convergence()
c1.plot_convergence_chi()

# Close the simulation renderer
c1.sim.close()

### What is the expected behavior?
I run with the same setup an EPR analysis which works just fine. The capacitative matrix should be 2*2 or more but it gives only a 1*1 matrix.

### Suggested solutions
I am sorry for my incompetence but it is my first project on Qiskit metal and it gives the same error for the trasmon cross. So in order to better understand how the simulation works I tried a simpler setup, but still no results.

SssZKing, 2024-12-06:
Do you have any qubit/transmon in your design? 

Based on the function `\qiskit-metal\qiskit_metal\analyses\quantization\lumped_capacitive.py:198, in extract_transmon_coupled_Noscillator(capMatrix, Ic, CJ, N, fb, fr, res_L4_corr, g_scale, print_info)` and its argument `capMatrix (np.ndarray): Order of the capacitance matrix must be bus1...busN-1, ground, Qubit_pad1, Qubit_pad2, readout. (in F)`

It looks to me that right now this function only supports pad-style transmon as I was trying to simulate a bare cross transmon and I got this shape error as well.

zlatko-minev, 2026-09-27:
Thanks for the report, and sorry for the slow response. Two things are going on here:

1. **The script as posted renders no qubit.** `q3d.render_design(['readout'], [])` renders only the readout meander, so Q3D returns a matrix with just ground/readout nets. `LOManalysis.run_lom()` expects a transmon in the matrix, ordered `bus1…busN-1, ground, pad1, pad2, readout`, with N = 1 + len(`c1.setup.freq_bus`). With the default `freq_bus=[6.0, 6.2]` that means 6×6. For a pocket transmon with only a readout coupler, render the qubit too and set `c1.setup.freq_bus = []`, which makes the expected size 4×4 (`ground, pad1, pad2, readout`).
2. **Cross transmons:** as @SssZKing noted, the legacy `run_lom()` only supports floating two-pad transmons. That's tracked in #1015, which also shows how to analyse a grounded crossmon with the newer `Cell` / `Subsystem` / `CompositeSystem` API (tutorials 4.04 and 4.05).

One more thing if you load a matrix yourself rather than running the Q3D simulation through `c1.sim`: there's currently a bug where setting only `c1.sim.capacitance_matrix` makes `run_lom()` fail with `KeyError: 0`. Until it's fixed, also set the per-pass data, in farads:

```python
c1.sim.capacitance_matrix = df                       # values in fF
c1.sim.capacitance_all_passes = {1: df.values * 1e-15}
```

Closing this as a duplicate of #1015 — please comment there if the new-LOM route doesn't work for your design.

---
_Generated by [Claude Code](https://claude.ai/code)_

closed 2026-09-27 by zlatko-minev