qiskit-community/qiskit-metal #1005: Issue in importing CompositeSystem module from qiskit_metal.analyses.quantization.lom_core_analysis module

vegaonline, 2024-12-27:
### Information

- **Qiskit Metal version**:0.1.5
- **Python version**: 3.12.4
- **Operating system**: Ubuntu 22.04 with kernel 6.8.0-50-generic

### What is the current behavior?
I was trying to run the a tutorial file "Analysis/B. Advanced - Direct use of the renderers/4.19 Analyze a transmon using ElmerFEM.ipynb".

Error: `ModuleNotFoundError: No module named 'qutip.qobj'` while IMPORTING from `qiskit_metal.analyses.quantization.lom_core_analysis`.  

### Steps to reproduce the problem
`from qiskit_metal.analyses.quantization.lom_core_analysis import CompositeSystem` generates the error everytime I use.

### What is the expected behavior?
There should not be error.

### Suggested solutions
I think there is some issue with `pyside2` (probably but not sure).

vegaonline, 2024-12-27:
I also think that it issue with lom_core_analysis as it is giving same error while importing Cell and Subsytem also.

zlatko-minev, 2024-12-28:
Do you have Qutip and tight version installed Sent from my iPhone

vegaonline, 2024-12-28:
Yes, I have Qutip inside `metal` environment.

vegaonline, 2024-12-29:
I had to re-install qiskit-metal afresh and then the issue was resolved. However, now that example runs and generates capacitance_matrix while similar code with different name generates error with 'nets'. The error says 'nets was not generated` for `elmer_renderer.add_solution_setup('capacitance')` and `elmer_renderer.run('capacitance')`.

vegaonline, 2025-01-03:
Latest Development:
I installed qiskit_metal afresh with pyside6. Now, adding of solver as 'capacitance' was successful and renderer started solving while suddenly code stopped for segmentation fault. Bbacktrace hinted issue at ELMERFEM/fem/src/modules/StatElecSolve.F90. The ElmerSolver log file is here 
[elmersolver.log](https://github.com/user-attachments/files/18301748/elmersolver.log) .

zlatko-minev, 2026-09-27:
Thanks for following up and for attaching the solver log. Here's where things stand on current `main` (quantum-metal 0.8.1).

**Import error (`qutip.qobj`):** as you found, this was an environment issue. On current releases, `from qiskit_metal.analyses.quantization.lom_core_analysis import CompositeSystem, Cell, Subsystem` imports cleanly with qutip 5.x.

**Segfault:** your log shows the model Metal generated was read without problems: 7 boundary conditions, 2 bodies and 5 capacitance bodies, which matches what the 4.19 tutorial design produces. The mesh loaded and the first electrostatic solve converged. The crash happens afterwards, inside ElmerSolver's multithreaded `GeneralElectricFlux` routine (`StatElecSolve.F90:989`), in your source build (rev `2d49aea29`, 8 OpenMP threads). That routine only computes the field/energy post-processing; by our reading of the Elmer source, the capacitance matrix doesn't depend on it. We don't have Elmer in our test setup, so we couldn't reproduce this. Two things worth trying (untested on our side):

1. Run single-threaded: set `OMP_NUM_THREADS=1` in the environment before starting Jupyter or Python.
2. Skip the field/energy step, before `add_solution_setup(...)`:
   ```python
   elmer_renderer.default_setup.capacitance.Calculate_Electric_Field = False
   elmer_renderer.default_setup.capacitance.Calculate_Electric_Energy = False
   ```
   You'll lose the field views in `display_post_processing_data()`.

Upstream Elmer has also reworked `StatElecSolve` since your build, so a newer Elmer may behave differently. If the crash persists with option 1 on a current Elmer build, it's most likely worth reporting to ElmerCSC/elmerfem with this log. Please let us know how it goes on 0.8.1. We're also going to make the renderer report solver crashes directly (it currently doesn't check ElmerSolver's exit status).

---
_Generated by [Claude Code](https://claude.ai/code)_

zlatko-minev, 2026-11-15:
Thanks @vegaonline for sticking with this and for attaching the solver log. On current
releases `from qiskit_metal.analyses.quantization.lom_core_analysis import CompositeSystem,
Cell, Subsystem` imports cleanly with qutip 5. On the Elmer side, v0.9.0 (#1187) checks
ElmerSolver's exit status and raises with the tail of its log. It also fixed the
`AttributeError: nets` you hit along the way (#1008). The segfault itself is inside ElmerSolver's
`StatElecSolve`, so if it persists with `OMP_NUM_THREADS=1` on a current Elmer build,
ElmerCSC/elmerfem is the place to report it. Closing.

closed 2026-11-15 by zlatko-minev