qiskit-community/qiskit-metal #1008: Capacitance body is not defined in Elmer rendering "add_solution_setup('capacitance')"

vegaonline, 2024-12-31:
### Information

- **Qiskit Metal version**: 0.1.5
- **Python version**:  3.11.11
- **Operating system**: Ubuntu 22.04 kernel 6.8.0-50-generic

### What is the current behavior?
For Elmer Rendering, rendering, meshing, gui launching is done without error while `elmer_renderer.add_solution_setup('capacitance')` tells 
(1) Capacitance body is not defined,
(2) No attribute "nets".

`
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[16], line 2
      1 # Add a solution to solve for Capacitance matrix
----> 2 elmerRenderer.add_solution_setup('capacitance')

File ~/WORK/QC/qiskit-metal/qiskit_metal/renderers/renderer_elmer/elmer_renderer.py:537, in QElmerRenderer.add_solution_setup(self, sim_name, solver_names, equation_name)
    534 bodies = self.define_bodies(setup, eqn, materials)
    536 # Add boundary conditions
--> 537 boundaries, cap_body = self.define_boundaries()
    539 if cap_body == 1:
    540     self.logger.warning(
    541         "WARNING: No capacitance bodies added to the model other "
    542         "than those connected to ground. ElmerFEM cannot run a "
    543         "capacitance extraction analysis without any bodies. ")

File ~/WORK/QC/qiskit-metal/qiskit_metal/renderers/renderer_elmer/elmer_renderer.py:619, in QElmerRenderer.define_boundaries(self)
    615             continue
    617     phys_grps_dict.update(ph_geoms)
--> 619 for net, geom_names in self.nets.items():
    620     geom_id_list = [
    621         phys_grps_dict[f"{name}_sfs"] for name in geom_names
    622     ]
    624     if net == "gnd":

AttributeError: 'QElmerRenderer' object has no attribute 'nets'
`

However, `elmer_renderer.py` has `assign_nets()`.

I also wish to mention that `Analyze_a_transmon_using_ElmerFEM.py` works smoothly without any error. 

### Steps to reproduce the problem

### What is the expected behavior?
It is expected to run that line without error.
[image]

### Suggested solutions

zlatko-minev, 2024-12-31:
Can you try it with the #1002  pull request and let @shanto268 there know if you still see it

vegaonline, 2024-12-31:
Thanks.
@zlatko-minev  and @shanto268, I installed pyside6 and tested. I am still getting those errors.  The issue creeps in here
`
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[20], line 2
      1 # Add a solution to solve for Capacitance matrix
----> 2 elmerRenderer.add_solution_setup('capacitance')

File ~/WORK/QC/qiskit-metal/qiskit_metal/renderers/renderer_elmer/elmer_renderer.py:537, in QElmerRenderer.add_solution_setup(self, sim_name, solver_names, equation_name)
    534 bodies = self.define_bodies(setup, eqn, materials)
    536 # Add boundary conditions
--> 537 boundaries, cap_body = self.define_boundaries()
    539 if cap_body == 1:
    540     self.logger.warning(
    541         "WARNING: No capacitance bodies added to the model other "
    542         "than those connected to ground. ElmerFEM cannot run a "
    543         "capacitance extraction analysis without any bodies. ")

File ~/WORK/QC/qiskit-metal/qiskit_metal/renderers/renderer_elmer/elmer_renderer.py:619, in QElmerRenderer.define_boundaries(self)
    615             continue
    617     phys_grps_dict.update(ph_geoms)
--> 619 for net, geom_names in self.nets.items():
    620     geom_id_list = [
    621         phys_grps_dict[f"{name}_sfs"] for name in geom_names
    622     ]
    624     if net == "gnd":

AttributeError: 'QElmerRenderer' object has no attribute 'nets'
`
Wish ALL of YOU a Happy and Prosperous NEW YEAR.

zlatko-minev, 2026-09-27:
Thanks for the detailed traceback, and sorry for the long delay.

We looked into this on current `main` (quantum-metal 0.8.1). `QElmerRenderer.nets` is only created by `QElmerRenderer.render_design(...)`, and both `add_solution_setup()` and `run()` depend on it. Your traceback shows the Gmsh geometry *had* been rendered, because `define_bodies()` got past `physical_groups`. That usually means the Elmer renderer object itself never ran `render_design`, e.g. the design was rendered through `elmerRenderer.gmsh.render_design(...)` or a separate `QGmshRenderer`. We reproduced exactly this `AttributeError` that way. With the order used in tutorial 4.19, the setup completes:

```python
elmer_renderer = QElmerRenderer(design)
elmer_renderer.render_design(open_pins=[...], skip_junctions=True)  # assigns nets
elmer_renderer.export_mesh()
elmer_renderer.add_solution_setup("capacitance")
elmer_renderer.run("capacitance")
```

The renderer should raise a clear error here instead of an `AttributeError`, and we'll add that guard. PR #1002 didn't change this code path, which is why testing it made no difference. If your script already calls `render_design` on the same `QElmerRenderer` object and still fails, could you share a minimal script on 0.8.1? We'll reopen then.

---
_Generated by [Claude Code](https://claude.ai/code)_

closed 2026-09-27 by zlatko-minev