qiskit-community/qiskit-metal #1010: Mixed route together with linetee throws error no attribute coords

SimonQuantumhw, 2025-01-17:
### Information

- **Qiskit Metal version**: newest
- **Python version**: idk
- **Operating system**: win11

### What is the current behavior?

ERROR [rebuild]: ERROR in building component name=line, error='NoneType' object has no attribute 'coords'

### Steps to reproduce the problem

Create mixedroute with anchors and between anchors and connect it to lineetee second_end

LineTee1 = LineTee(
design,
name='Lt1',
options={'pos_x': '500.0um'},component_template={'falseparam1': {'falseparam2': 'false-param', 'falseparam3': 'false-param'}},
)LineTee1 = LineTee(
design,
name='Lt2',
options={'pos_x': '-500.0um',},component_template={'falseparam1': {'falseparam2': 'false-param', 'falseparam3': 'false-param'}},
)LineTee1 = LineTee(
design,
name='Lt3',
options={'orientation': '90.0',
 'pos_x': '-2500.0um'},component_template={'falseparam1': {'falseparam2': 'false-param', 'falseparam3': 'false-param'}},
)LineTee1 = LineTee(
design,
name='Lt4',
options={'orientation': '270.0',
 'pos_x': '2500.0um'},component_template={'falseparam1': {'falseparam2': 'false-param', 'falseparam3': 'false-param'}},
)from qiskit_metal.qlibrary.tlines.mixed_path import RouteMixed
import numpy as np
from collections import OrderedDict
ops=dict(fillet='90um')anchors = OrderedDict()
anchors[0] = np.array([1.5, 1.])
anchors[1] = np.array([0, 2])
anchors[2] = np.array([3., 3])between_anchors = OrderedDict() # S, M, PF
between_anchors[0] = "S"
between_anchors[1] = "M"
between_anchors[2] = "S"
between_anchors[3] = "S"optionsR = {'pin_inputs': {
              'start_pin': {'component': 'Lt2', 'pin': 'second_end'},
              'end_pin': {'component': 'Lt3', 'pin': 'second_end'}
            },
            'total_length': '2mm',
            'chip': 'main',
            'layer': '1',
            'trace_width': 'cpw_width',
            'step_size': '0.25mm',
            'anchors': anchors,
            'between_anchors': between_anchors,
            'advanced': {'avoid_collision': 'true'},
            'meander': {
              'spacing': '200um',
              'asymmetry': '0um'
            },
            'snap': 'true',
            'lead': {
              'start_straight': '0.3mm',
              'end_straight': '0.3mm',
            },
          }qa = RouteMixed(design, 'line', optionsR)gui.rebuild()
gui.autoscale()

### What is the expected behavior?

It should create route

### Suggested solutions

zlatko-minev, 2025-01-26:
Have you had a chance to try to see where the error might be and to correct, we can include in v0.5, or get some help in #1002

mvanhorn, 2026-10-04:
RouteAnchors.unobstructed_close_up wrapped the unary union of the obstacle polygons and buffered paths in a GeoSeries and read geometry.exterior[0].coords. GeoSeries.exterior is None when that union is a MultiPolygon or GeometryCollection, which is what disjoint buffered paths on a tee produce, so the .coords access raises AttributeError.

unobstructed_close_up now flattens the unary union into Polygon pieces, including geometries nested in a MultiPolygon or GeometryCollection, and tests the segment against each exterior ring.

closed 2026-10-11 by zlatko-minev