qiskit-community/qiskit-metal #1013: It works but suddenly close after redenring GUI

dgks0n, 2025-01-22:
Hi there~ I tried to make a sample with `qiskit-metal` version 0.1.5 with below steps.
I created simple python file name as `helloworld.py` with simple snippets:

```
from qiskit_metal import MetalGUI
from qiskit_metal import designs

design = designs.DesignPlanar()
design.overwrite_enabled = True

design.chips.main.size.size_x = '11mm'
design.chips.main.size.size_y = '9mm'

gui = MetalGUI(design)
```

Finally, I executed that python scripts by using command `python helloworld.py`, it works but immediately close GUI after that. I even cannot see the GUI and there is no error logs.

Strange thing is, it works as well with above code snippets with Jupyter notebook. I am wondering why?
How can I solve this problem?

zlatko-minev, 2025-01-22:
If you’re running from a script, it might be closing the event loop in QT, the gui framework we use. In Jupiter, the way we automatically set it up is that there is a background asynchronous event that is running and keeping the GUI open. There is a function somewhere in the documentation or if you just google it or ask some AI agent to tell you how to run it from a script so that the event loop runs in the background or blocks the executionSent from my iPhone

SamWolski, 2025-02-11:
@zlatko-minev I think this is the issue I brought up on the Slack a while back.

@dgks0n the workaround is to add a call to
`gui.qApp.exec_()`
in the script, which is a blocking call that should make the GUI behave. Be aware that it will cause a segfault after the window is closed, so is only suitable for the end of the script after all generation functions have been run.