qiskit-community/qiskit-metal #1014: Legacy editable install deprecation

SamWolski, 2025-01-29:
### Information

- **Qiskit Metal version**: 0.1.5
- **Python version**: 3.10.16
- **Operating system**: macOS Sequoia 15.3

### What is the current behavior?

Attempting to perform an editable pip install from a local file results in a deprecation warning from pip/setuptools.

```
DEPRECATION: Legacy editable install of qiskit_metal==0.1.5 from file://<...>/qiskit-metal (setup.py develop) is deprecated. pip 25.1 will enforce this behaviour change. A possible replacement is to add a pyproject.toml or enable --use-pep517, and use setuptools >= 64. If the resulting installation is not behaving as expected, try using --config-settings editable_mode=compat. Please consult the setuptools documentation for more information. Discussion can be found at https://github.com/pypa/pip/issues/11457
```

### Steps to reproduce the problem

1. Clone the qiskit-metal repository to a local folder
2. Attempt to install using `pip install -e <path-to-repo>`

### What is the expected behavior?

Setup definitions for the qiskit-metal package should not trigger the deprecation warning.

### Suggested solutions

Define package setup using pyproject.toml instead of setup.py

zlatko-minev, 2025-01-31:
Let’s see if we have addressed this in pull request #1002, any help is appreciated Sent from my iPhone

SamWolski, 2025-01-31:
It is not currently addressed; the same message pops up on the PR fork.

abhishekchak52, 2025-12-14:
Should be fixed by #1026. `uv_build` replaced `setuptools` as the build backend and `setup.py` was replaced by `pyproject.toml`

Closing as completed.