qiskit-community/qiskit-metal #1019: Issue with Ansys2025 R1: different results in eigenmode and EPR analysis tutorial 4.02

GuidoMenichetti, 2025-05-06:
### Information

- **Qiskit Metal version**:  0.1.5 0.1.5
- **Python version**: 3.11
- **Operating system**: Windows

### What is the current behavior?

We are encountering an issue while running the Qiskit Metal tutorial 4.0.2: Eigenmode and EPR analysis. Specifically, the results we obtain do not match those reported in the tutorial.

    The following assignment in the code does not seem to update correctly:

    eig_qb.sim.setup.vars.Lj = '11 nH'

    When we check directly in Ansys Electronics Desktop (version 2025 R1), the value of Lj remains at '10 nH', the original default.

    As a result, the 6th step value of eig_qb.sim.convergence_f is 6.20 rather than 6.25 as shown in the tutorial.

We suspect the issue might be related to the interface between Qiskit Metal and Ansys. We used the recommended Qiskit environment for the installation.

Steps to Reproduce:

    Follow tutorial 4.0.2 as provided.

    Attempt to set eig_qb.sim.setup.vars.Lj = '11 nH'.

    Check the value of Lj inside Ansys — it remains '10 nH'.

    Observe the different output in eig_qb.sim.convergence_f.

Expected Behavior:
The updated value for Lj should be correctly passed to Ansys, reflecting '11 nH', and the output should match the tutorial's results.

Could this be a compatibility issue with Ansys 2025 R1, or is there a recommended workaround?

Thanks in advance for your support!

### Steps to reproduce the problem

### What is the expected behavior?

### Suggested solutions

zlatko-minev, 2026-09-27:
Thanks @abhishekchak52 — I can confirm your diagnosis from the code:
- `sim.setup.vars` (e.g. `Lj = "11 nH"`) is only written to Ansys as a design variable (`QAnsysRenderer.set_variables`).
- The junction's lumped RLC boundary takes its inductance from the component's `hfss_inductance` option, which defaults to the literal `"10nH"`. So unless the qubit is told to use the variable, HFSS simulates 10 nH, which is what @GuidoMenichetti saw in AEDT.
- pyEPR reads `Lj` from the design variable for the quantization step, so the EPR Hamiltonian uses 11 nH while the field simulation used 10 nH.

This isn't new (0.1.5 had the same default) and isn't specific to Ansys 2025. The workaround:

```python
design.components["Q1"].options["hfss_inductance"] = "Lj"
design.components["Q1"].options["hfss_capacitance"] = "Cj"
design.rebuild()
```

Run it before `eig_qb.sim.run(...)`. Two changes make this harder to miss:
- #1184 adds that step, with an explanation, to section I of tutorial 4.02.
- #1186 makes `EigenmodeSim` warn before rendering whenever `setup.vars.Lj` differs from a junction's fixed inductance.

On the 6.25 vs 6.20 GHz difference: a larger Lj lowers the frequency, so the 11 nH vs 10 nH mix-up alone doesn't explain the tutorial showing the *higher* value. It's more likely an Ansys version or meshing difference, but we can't confirm that without re-running on Ansys. If someone re-runs the updated tutorial on 2025 R1/R2 and posts the convergence table, that would settle it.

---
_Generated by [Claude Code](https://claude.ai/code)_

zlatko-minev, 2026-11-15:
Thanks @GuidoMenichetti for spotting that `Lj` never reached AEDT, and @abhishekchak52
for tracing it to the junction's `hfss_inductance`. Both parts shipped in v0.9.0. Tutorial
4.02 now links the junction to the `Lj`/`Cj` design variables (#1184), and `EigenmodeSim`
warns when `setup.vars.Lj` differs from a junction's fixed inductance (#1186). Closing. The
remaining 6.25 vs 6.20 GHz gap goes the opposite way from the Lj mix-up, so it is probably
an AEDT-version or meshing difference. If you re-run the updated 4.02 on 2025 R1/R2, a
convergence table posted here would settle it.

closed 2026-11-15 by zlatko-minev